UnchartedBull/OctoDash #3760: Please specify OctoPrints full virtualenv path manually (no trailing slash

mtscott42, 2023-07-12:
**What doesn't work?**
When running the install script I keep getting this message and don't know how to get past it.

"Please specify OctoPrints full virtualenv path manually (no trailing slash"

**What did you already try?**
Searching the web to find solutions with no luck.  

**General Information:**
 - Hardware PI 4B 
 - OS Info Octoprint 
 - OctoDash Version
 - OctoPrint Version 1.9.1

Thanks much!

riddochn, 2023-07-18:
My problem was that I had installed raspian using something different from the defualt "pi" user name. I had to enter the location as /home/username/oprint

Extra-Fox, 2023-12-27:
I forget this every time myself, it would be nice if a little text was added that said something to the effect (hey, did you create a different user for OctoPrint?)

quickllama, 2024-01-18:
> My problem was that I had installed raspian using something different from the defualt "pi" user name. I had to enter the location as /home/username/oprint

Exactly the explanation and fix I needed. Thanks!

zandr, 2024-02-20:
The install script has `/home/pi` hard-coded where it should be using $HOME. I just did a fresh install after fixing the hardcoding, and it installed fine. I'll write a PR after I get a cup of coffee.