UnchartedBull/OctoDash #3818: Waveshare 5" and 4.3" display not working

derVicthor, 2023-08-11:
**What doesn't work?**
When setting up an OctoPi installation and adding OctoDash, either the touch isn't working or the screen is shown with a black border on the right and touch is way off.

**What did you already try?**
Changes config.txt configuration and contacted Waveshare support.

**General Information:**
Raspberry Pi 3B+ with 800x600
Linux octopi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
OctoPrint 1.9.2
OctoDash v2.3.1, armv7

**Additional context**
Tried the following configuration:

config.txt
dtoverlay=vc4-kms-v3d --> full screen, but no touch
dtoverlay=vc4-fkms-v3d --> black border on the right, but touch works (although off center)

derVicthor, 2023-08-12:
As a heads up: disabling the driver completely in the config.txt fixed the issue.

closed 2023-08-12 by derVicthor

ichibana, 2023-11-03:
Hi there! I'm also unable to get touch to work on my Waveshare 4.3. Tracking that in [this other thread](https://github.com/UnchartedBull/OctoDash/issues/3938).

I tried your approach and it didn't work, unfortunately. Do you have any other potential solutions?

ichibana, 2023-11-04:
Oops! I missed that there was another driver call earlier in the file. Commented that out too and it works now! Thanks for the suggestion.

...do you know why this fixes the problem?