UnchartedBull/OctoDash #3819: After changing user in OctoPrint: Not connecting anymore

EmEiBee, 2023-08-13:
After changing the administrative user in OctoPrint, Octodash no longer connects.
Error message:
Initializing takes longer than expected. Please make sure OctoPrint is running and CORS is enabled for the API.

OctoPrint is running. It is easy to work and print via the WebGUI.

I blindly searched various files because I suspect that the legitimate user is stored somewhere in OctoDash, which OctoDash can no longer find.

OctoPrint = 1.9.1, OctoDash = 0.0.7 on RaspBerryPI 3B
7" touchscreen on it

Greetz
Micha

tomtips57, 2024-06-25:
Did you ever resolve this problem?  I am have the same issue and would appreciate some assistance.  Thank you!

EmEiBee, 2024-06-25:
Nope, sorry. I stopped using Octo for months and switched to Repetier...