UnchartedBull/OctoDash #3820: OctoDash install errors due to architecture mismatch

BlackFire2357, 2023-08-13:
**What doesn't work?**
I am re-installing OctoDash on my RPi4B with Miuezi 3.5" touchscreen.   During the install process (both manual and automatic), the following lines appear:
    dpkg: error processing archive octodash.deb (--install):
     package architecture (arm64) does not match system (armhf)
    Errors were encountered while processing:
     octodash.deb

The installer then continues to ask to setup OctoDash to automatically start on boot and I can finish the install steps.

Upon boot, the Pi will display the four raspberry logos at the top of the screen, text scrolls by until the screen goes completely black with a white underline cursor in the upper left corner.

OctoPrint web interface works and controls the printer as expected.

**What did you already try?**
Fresh install of OctoPi/OctoPrint from the Raspberry Pi Imager
Installed display drivers
Ran OctoDash install script: bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
Every troubleshooting step provided in the guide (https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#installation)
In the OctoPrint web interface, selected "Start OctoDash" from power menu; no reaction.
In the OctoPrint web interface, selected "Restart OctoDash" from power menu; no reaction.

**General Information:**
 - OctoPi 1.0.0, running on Raspberry Pi 4 Model B Rev 1.5
 - OctoPrint 1.9.2
 - Miuzei 4" touchscreen (https://www.amazon.com/gp/product/B07XBVF1C9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

**Additional context**
This is my fourth instance of this same hardware setup, so I know it works.  And this exact hardware worked before I re-installed OctoPi.
Copying the SD card from an existing setup does not work; the PI will not boot with the copied files.
I am a toddler when it comes to OctoPi language; Point and Click instructions are necessary.

jneilliii, 2023-08-13:
This is kind of a known issue with specific versions of Bullseye/OctoPi image. Raspberry Pi changes the architecture for no good reason while performing apt update/upgrade. The workaround is to manually download and install the armv7 build and install it with the following commands.

```
wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_armv7l.deb
sudo dpkg -i octodash.deb
```
reboot the pi and in theory you should be in business.

BlackFire2357, 2023-08-14:
No change.
The package downloaded (did not get a copy of the text before the SSH session closed), and installed:

_pi@octopi:~ $ sudo dpkg -i octodash.deb
[sudo] password for pi:
(Reading database ... 97052 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (2.3.1) over (2.3.1) ...
Setting up octodash (2.3.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for mailcap (3.69) ...
pi@octopi:~ $_

Rebooted and it comes back to the same black screen with white cursor.  Starting and restarting OctoDash via the Pi's web interface makes no changes.

RedlineMods, 2024-01-29:
I am running the exact same setup as you.  Raspberry Pi4 with Miuzei 4" Touchscreen.  After a lot of trail and error.  Here is what I did exactly to get everything up and running.  You will probably need to start from scratch as I had to do this multiple times to get it up and running.  Following the instructions below and you should be good to go.

Raspiberry Pi Imager Octoprint.
SSH using username and Password
(copy and paste the below one line at a time)
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MPI4008-show
sudo reboot
sudo apt-get install git build-essential xorg-dev xutils-dev x11proto-dri2-dev
sudo reboot
sudo apt-get install libltdl-dev libtool automake libdrm-dev
sudo reboot
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
/home/rm/oprint
sudo raspi-config
(This section is done with arrow keys)
[1 System Options -> S5 Boot / Auto Login -> B2 Console Autologin
3 Boot Options -> B2 Console Autologin
<Finish> and <Yes>]
(Back to copy and paste one line at a time.)
wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.0.0/octodash_2.0.0_armv7l.deb
sudo dpkg -i octodash.deb
sudo reboot
wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_armv7l.deb
sudo dpkg -i octodash.deb

queengooborg, 2025-03-18:
This should now be fixed following some updates to the install script.  If it's still not working, let us know!

closed 2025-03-18 by queengooborg