UnchartedBull/OctoDash #3821: Is it my screen that is the issue?

Xlr8Bigmike, 2023-08-14:
**What doesn't work?**
The touch screen works hit or miss, and then sometimes just freezes up to where i have to reboot octodash.

**What did you already try?**
honestly not sure what to try. 

**General Information:**
Pi 3b+
 Touchscreen-Smraza 5" Inch https://www.amazon.com/gp/product/B095HQPY5X/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
it is a screen i got a couple years ago, and they don't sell it anymore. so maybe its no good?

I just installed the latest version of octopi and i believe octodash

im assuming by running this (bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)) that would install the latest version correct?

**Additional context**
I am using a longer DSI cable maybe 2' could that be the issue?  

Thanks for the help,
Mike

Xlr8Bigmike, 2023-08-14:
I got a shorter dsi cord and it solved my issue i believe!

EmEiBee, 2023-08-28:
Same here with an "ELECROW 5 Zoll Raspberry Pi Display Touchscreen Monitor 800x480 TFT LCD Display"
Also scrolling (https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions / more than 6 actions) don't work.

Hillshum, 2025-11-23:
I'm going to close this as it looks like the original issue was solved. If you are still having issues please open a new ticket with details of your setup.