UnchartedBull/OctoDash #3881: OctoDash wont rotate.

EliasVP, 2023-09-15:
Hello, 

I'm running OctoPrint and OctoDash on a RPI 4 and all is working fine exept that OctoDash just wont go in Landscape mode. 
I added the rotate=270 to the end of this line in config.txt.

dtoverlay=vc4-kms-dpi-hyperpixel4sq,rotate=270

and that works just fine for the booting screen but once OctoDash starts it goes back to portrait. what did work was when I tried to rotate it 180°, then Ocodash did launch upside down in portait, but nothing when I put it to 90 or 270. Next I also tried adding

dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x

but that did just the same. So i tried putting

display_lcd_rotate=3

to the end of the config.txt file but that doesn't seem to do anything.

So I don't know what else I could try.

TheLivingBubba, 2023-09-16:
Have you tried just lcd_rotate=2 into config.txt?

EliasVP, 2023-09-16:
Yes, I tried it with the "rotate=270" at the end of the other line and without it.

EliasVP, 2023-09-16:
I have found the solution in the comments on a simular problem. https://github.com/pimoroni/hyperpixel4/issues/177#issuecomment-1237236115

So I just have to have 

dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x

in the config.txt. and make an extra file in this place 

/usr/share/X11/xorg.conf.d/90-monitor.conf

and put 

Section "Monitor"
    Identifier "DPI-1"
    Option "Rotate" "right"
EndSection

in the file and then everything works like a charm (just swap "right" for "left" to rotate 270° in stead of 90).

closed 2023-09-16 by EliasVP