UnchartedBull/OctoDash #3931: Stuck at “Hey there” screen

Schalk-cas12, 2023-10-13:
Hey I just installed octodash for the first time and I am having some trouble.  Octodash installed fine and boots up to the "Hey there" screen but for whatever reason my touch screen wont register inputs. I've used this exact screen in RPI OS and it picks up inputs fine. Any ideas on what I am doing wrong?

borderline667, 2023-10-14:
Following. Same thing for me. What are the screen you are using?

jhollandb, 2023-10-14:
I'm having the same issue.  I am using a Waveshare 5-inch DSI LCD.  I have input the required text into the config.txt file.  I was able to use the screen about five months ago before I moved.

Schalk-cas12, 2023-10-15:
> Following. Same thing for me. What are the screen you are using?

I am using the FREENOVE 5 inch touchscreen

Schalk-cas12, 2023-10-15:
> I'm having the same issue.  I am using a Waveshare 5-inch DSI LCD.  I tested to see if the screen was the problem by installing another touch OS.  It worked flawlessly.
> 
> 
> 
> I have input the required text into the config.txt file for my screen. I was able to use the screen about five months ago before I moved.  

What edits did you make to your config.txt , did you follow a guide?

jhollandb, 2023-10-15:
> What edits did you make to your config.txt , did you follow a guide?

I followed the Waveshare Wiki. https://www.waveshare.com/wiki/5inch_DSI_LCD

added to config.txt at the end.

dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch

Schalk-cas12, 2023-10-16:
> I followed the Waveshare Wiki. https://www.waveshare.com/wiki/5inch_DSI_LCD
>
>
>
> added to config.txt at the end.
>
>
>
> dtoverlay=vc4-kms-v3d
>
> dtoverlay=vc4-kms-dsi-7inch

Ok thanks I'll try that

borderline667, 2023-10-16:
> Ok thanks I'll try that

You may need to use your own config for your device, I am on that troubleshooting I'll let you know if I find the solution 👍