UnchartedBull/OctoDash #3933: OctoDash rotation on Pi4b and Hyperpixel4 rect.

G0rki1, 2023-10-15:
**What doesn't work?**
I can't get running OctoDash in right orientation. Either I use the legacy drivers and can rotate the display correctly but not the touch orientation or I use the standard drivers from OctoPi Image 1.93 and manage to rotate the touch but not the display

**What did you already try?**
I try to find a solution for some days now.
I already tried the solution von hyperpixel-support (issue#177) and manage to rotate the display 90° clockwise with legacy drivers installed. In this config I can't get the touch rotated 90°.
I also tried to use the standard driver coming with the image an just adjusted config.txt with the following lines
```
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```
With this settings the starting sequence of Pi starts in correct orientation and touch works correct, but the orientiation of OctoDash is eigher 90° left or 90° right (portrait) but not landscape modes.

**General Information:**
 - Raspi Pi 4b 2G
 - OctoPi 1.0.0 from Raspi Pi Imager 1.75
 - OctoDash 2.3.1
 - OctoPrint 1.9.3

G0rki1, 2023-10-22:
Found my solution

closed 2023-10-22 by G0rki1

MrMontesa, 2023-12-29:
Hey @G0rki1 have you found a solution to this issue? Currently facing exactly the same. After updating /boot/config.txt as per new style, the boot screen is in correct horizontal orientation. As well as the touch. But as soon as OctoDash starts, its back in vertical orientation.

G0rki1, 2023-12-29:
HI, yes, I found a solution. I used the legacy driver and roughly followed these instructions:
[](https://kevenaar.name/octopi-with-octodash-on-rpi-4-with-hyperpixel-4-0/)
I combined this with this guide:
[](https://www.instructables.com/Rotate-Raspberry-Pi-Display-and-Touchscreen/)
When editing with the command
`sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf`
I originally made the mistake of starting the line with a space, instead you have to use TAB.
Somehow it worked... :-)