UnchartedBull/OctoDash #3934: OctoDash not wanting to install on setup on PI

EricSV06, 2023-10-16:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.
i'm using Pi 4 B 
OctoPrint 1.9.3
OctoPi 1.0.0
when i type in the Bash code to get it. it only goes to the next promt line.... doesnt say anything. restart and nothing. only octopi and print are on it.....

MaurolepisDreki, 2023-10-27:
Just finished troubleshooting a similar problem on a friend's Pi, and the trouble is in the install and update scripts (lines 671 and 3 respectively) in how the script detects architectures.

I'll bet that if you run `dpkg --print-architectures` that it will tell you it is running armhf rather than aarch64?
The immediate solution is (then) to download and install the armv7l.deb file instead from https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_armv7l.deb as described at https://github.com/UnchartedBull/OctoDash/wiki/Installation#manual-installation.

The long term solution is to replace the architecture detection mechanism in the install and update scripts from `uname -m` to `dpkg --print-architectures`.  But I dare not make that patch myself because I (personally) have no way of debugging such a patch.
This is not an uncommon problem, and I can only imagine how many rage-quits it causes.

EricSV06, 2023-10-27:
I got it working finally! Many attempts after 🤭🤷‍♂️ darn things. Thanks
though. Now I'll know if I ever run into it again or a buddy.

jneilliii, 2023-10-27:
I have a patched version of the script that should help in this situation. The command to run would be this...

```
bash <(wget -qO- https://raw.githubusercontent.com/jneilliii/OctoDash/bugfix/arch_detection/scripts/install.sh)
```

Only change to the script is how the architecture is detected using dpkg instead of uname as seen [here](https://github.com/jneilliii/OctoDash/commit/90e80ab17754fd55a873c38ed1a45623008518cf).

zrosgit, 2024-03-24:
Thank you @jneilliii .  The URL in your post isn't valid, but the install.sh script worked for me on my RPi4 with a Hosyond 3.5 tft (gpio) screen.

jneilliii, 2024-03-24:
ah, I merged the changes into my main branch...here's the updated command line.

```
bash <(wget -qO- https://raw.githubusercontent.com/jneilliii/OctoDash/main/scripts/install.sh)
```

zrosgit, 2024-03-24:
Great.  I verified that this URL did do the wget operation without issue.  Before I found your script, I must have spent a minimum of three days trying to get OctoDash to install and run.  Was about to give up. As I understand it, using the armv71.deb (your script for my Pi) vs the arm64.deb (which is what  @UnchartedBull's install script was selecting for my pi) made the difference.

jneilliii, 2024-03-24:
Yeah, more than likely. I also just made an adjustment this morning from #4152 that allows installing on installs that aren't using the username pi.

closed 2025-03-18 by queengooborg