UnchartedBull/OctoDash #3938: Waveshare 4.3 DSI lcd screen: can't get the touchscreen to work with octodash

borderline667, 2023-10-17:
**What doesn't work?**
After the install. Octodash is showing up correctly, but I can't use my touchscreen it does not work, so I'm stuck at the HEY THERE welcome screen.

**What did you already try?**
Tried 4-5 times to reinstall everything. Touchscreen is working I could test it. There is probably a stupid thing that is missing in my config.txt file :-(

**General Information:**
 - Hardware raspberry pi 4a+
 - Octopi installed per Chris how-to  https://www.youtube.com/watch?v=_MJLJz-pJAA&t=1269s
 - OctoDash lastest Version 
 - OctoPrint Version 1.9.3

Here is my config.txt file. If someone could help I would reallly appreciate!

arm_64bit=0
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
display_rotate=2
#max_usb_current=1
hdmi_group=2
#hdmi_mode=1
hdmi_mode=87
#hdmi_cvt 800 533 60 6 0 0 0
hdmi_timings= 800 533 0 1 41 2 272 0 2 10 2 0 0 0 60 0 9009000 3
#dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
#hdmi_drive=2
hdmi_drive=1
hdmi_force_hotplug=1
#max_usb_current=1
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
# Automatically load overlays for detected cameras
camera_auto_detect=0

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
# enable raspicam
start_x=1
gpu_mem=128
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch

ichibana, 2023-11-03:
I also have this problem, for what it's worth. I tried commenting out the drivers per [this post](https://github.com/UnchartedBull/OctoDash/issues/3818), but that didn't work either.

ichibana, 2023-11-03:
I also re-imaged the SD card back to a stock Raspberry Pi OS (not OctoPi) and confirmed the touchscreen works and responds to input in that state. So the hardware is known good.

borderline667, 2023-11-04:
ichibana: This finally worked for me!
Remove this from your config file: 
**dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch**
 and add this line: 
**_dtoverlay=ft5406_** 

So basically, the steps are:

1-Install a fresh and clean OCTOPI
2-sudo apt update
3-sudo apt install -f
4-sudo apt upgrade
5-add this to your config file: dtoverlay=ft5406 and comment all other "dtoverlay" lines
6- Install Octodash

Let me know if this worked for you

ichibana, 2023-11-04:
Trying now, @borderline667 ! I did come across [this post](https://forums.raspberrypi.com/viewtopic.php?t=323496) which said something about that command being... out of date? I'm no expert, so read the post in that thread if you're curious. It's near the bottom.

That being said... I'll give it a whirl right now.

borderline667, 2023-11-04:
> Trying now, @borderline667 ! I did come across [this post](https://forums.raspberrypi.com/viewtopic.php?t=323496) which said something about that command being... out of date? I'm no expert, so read [...]

This is Interesting! I'm no expert as well and this screen gave me a lot of headaches. I now have wifi connectivity issue but the screen is still working well. It's still a work in progress :-)

ichibana, 2023-11-04:
@borderline667 where did you learn this arcane wisdom? Would love to read the source and familiarize myself with the context too.

ichibana, 2023-11-04:
Hey @borderline667, you know what ended up working? Commenting out **all** instances of **dtoverlay=vc4-kms-v3d
** driver. I thought I had earlier, but there was a line a bit earlier in the config.txt that also had this call. This is what was suggested in [this thread](https://github.com/UnchartedBull/OctoDash/issues/3818), so I'll go back over there and comment accordingly too. I did **not** utilize the **dtoverlay=ft5406** call. I would be curious if you can also achieve success without this line if you comment out the earlier dtoverlay text that is in the config file.

If anyone reading this can help me learn why removing the Waveshare-recommended drivers is what **fixes** the issue... I'd love to learn.

borderline667, 2023-11-04:
> Hey @borderline667, you know what ended up working? Commenting out **all** instances of **dtoverlay=vc4-kms-v3d ** driver. I thought I had earlier, but there was a line a bit earlier in the [...]

I'll try to do this in the next days. Not sure where I found that call, maybe here: [https://community.octoprint.org/t/octoprint-octoscreen-touch-not-working/52493](url) I need to learn more on how linux commands work I guess

ichibana, 2023-11-04:
Thanks for sharing the source! I’ll give it a read. 

Are you having WiFi problems with octodash only or also your octoprint setup entirely?

borderline667, 2023-11-04:
Only with octodash. In the meantime I have switched for another router (eero 6+) so I'll give another try to see if it fix my new issue. But octoprint itself work flawlessly