UnchartedBull/OctoDash #4128: Display IP of octoprint.

JamesM92, 2024-02-05:
**Is your feature request related to a problem? Please describe.**
I have a octopi at my employer with octodash, but if the IP address changes there is nothing on the dash to tell me what the IP address is, and octodash took over the screen quickly making it hard to catch the IP address. 

**Describe the solution you'd like**
Give a location in the settings menu to show what the current IP address of the octopi is encase the IP has changed regardless of any changes.

lathemaster, 2024-09-05:
I would like this option as well - display the IP address bottom center between Machine Name and Status

driftfox, 2024-12-15:
I was looking for this option in OctoDash today and realized it wasn't in there. I just created a PR that will hopefully be approved to add this to the About screen.

Sample screenshot:
[image]

queengooborg, 2025-03-16:
This has been fixed by #4633!

closed 2025-03-16 by queengooborg