UnchartedBull/OctoDash #4288: touchscreen not working

Nicoduino, 2024-05-17:
Hello, I just finished installing OctoDash but my touchscreens aren't working... I triend with the official one and a HDMI one... Please Help

queengooborg, 2025-03-18:
Can you please describe exactly how the touchscreens aren't working?  The more detail, the better, especially details like what specific screen you're using, what Raspberry Pi, which distro and version, etc.

Nicoduino, 2025-03-18:
Thanks for this reply. I was using waveshare's raspberry pi screen (it works like the official one) and an other HDMI screen m. The raspberry pi was a 4b with 2gigs of ram. I was using raspberry pi os lite as the distro. Since I got a bambulab, I dont need octodash anymore but I will maybe try to link another printer.

Invarti, 2025-05-20:
Might be the same issue as was flagged here: https://github.com/UnchartedBull/OctoDash/issues/3707

Try setting the correct device driver and see if that resolves the problem.