UnchartedBull/OctoDash #4292: Can't get the OctoDash screen rotated on Hyperpixel Touchscreen

travmilne, 2024-05-21:
I've tried all of the solutions I've come across online and nothing is getting the OctoDash orientation to rotate. The Raspberry Pi itself is rotated just fine, but OctoDash changes back to portrait mode once it loads.

Using a rectangular Hyperpixel monitor on a Raspberry Pi 4B.

Willing to do a fresh install and start over, but need to know how to actually get this working in 2024.

Should I just install with legacy drivers?

Why does everything else rotate properly but OctoDash is stuck in portrait mode?

travmilne, 2024-05-21:
Finally got this to work after countless attempts.

I did a fresh install of Raspberry Pi + OctoPrint via the Imager.

Instructions here under "Installing OctoPi using the Raspberry Pi Imager"
https://octoprint.org/download/

Then I edited the /boot/config.txt file to include:

dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x

Then I edited /usr/share/X11/xorg.conf.d/90-monitor.conf file to include:

Section "Monitor"
Identifier "DPI-1"
Option "Rotate" "left"
EndSection

Saved the file, reboot, and it boots directly into OctoDash with the landscape orientation.

You can change "left" to "right" to get it to flip in the other direction.

Finally got it figured out after hours of reinstalling and trying different solutions.

Have to give [this comment](https://github.com/UnchartedBull/OctoDash/issues/3881#issuecomment-1722254202) a bunch of credit because this was the step that finally go it to work.

morphias2004, 2024-05-22:
Just follow the guide in the Wiki, which I wrote; and you won't have issues.

I literally just used it 2 days ago because I hadn't run up OctoPi in ~6 months and needed to refresh my memory.

morphias2004, 2024-05-22:
@travmilne just follow the guide in the Wiki, which I wrote; and you won't have issues.

https://github.com/UnchartedBull/OctoDash/wiki/HyperPixel-4

I literally just used it 2 days ago because I hadn't run up OctoPi in ~6 months and needed to refresh my memory.

cdrsteve, 2025-01-05:
Finally....a rotation solution that works!!   Thank you, thank you, thank you.  Banging head against wall and was about to give up after untold hours of trying "solutions" that never seem to work.

closed 2025-03-18 by queengooborg