UnchartedBull/OctoDash #4293: Setting extruder temp sets temp wrong extruder for Prusa XL

morphias2004, 2024-05-22:
**Describe the bug**
When you set the extruder temp via the main screen on OctoDash, it sets the temp on Tool 1 instead of the currently loaded tool.

**To Reproduce**
Steps to reproduce the behavior:

1. Pick up any tool other than Tool 1
2. Set the extruder temp via OctoDash touch interface
3. Watch Tool 1 extruder temp rise and nothing happen on the currently loaded tool.

**Expected behavior**
The command should go to the currently loaded tool.

**General Information:**
 - RPi 4 4GB    Prusa XL 5 tool head on FW 6.0.1
 - OctoPi/ OctoPrint v 1.10.1
 - OctoDash Version [v2.3.1

morphias2004, 2024-05-22:
Working with a friend, he has now opened this directly related ticket for OctoPrint:

https://github.com/OctoPrint/OctoPrint/issues/5018

Hillshum, 2025-03-21:
v2.4.1 of OctoDash has support for multiple tools (with shared nozzle support coming shortly in v2.4.2). Can you please install that and test?