UnchartedBull/OctoDash #4313: -bash: cd: /usr/share/X11/xorg.conf.d: No such file or directory

patritha, 2024-05-30:
**What doesn't work?**
i want to follow the hyperpixel4 guide but the step there it says to type `cd /usr/share/X11/xorg.conf.d` it says the directory dosnt exist. while the directory dosnt exist, i followed the instructions exactly and dont understand why it wouldnt.

**What did you already try?**
im unsure of what do do next and in a attempt to not break anything ive just stopped and googled it, only to find 0 results

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 -      raspberry pi 4b
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 -      octopi v1.0.0
 - OctoDash Version [e.g. v1.0.0]
 -      havent gotten there yet
 - OctoPrint Version [e.g. v1.0.0]
 -      octoprint v1.10.1

**Additional context**
n/a

jneilliii, 2024-05-30:
It may not exist if you haven't installed octodash yet.

patritha, 2024-05-30:
youre right thank you!

closed 2024-05-30 by patritha