UnchartedBull/OctoDash #4314: black screen hyperpixel4

patritha, 2024-05-31:
**What doesn't work?**
the screen displays nothing when given power. 
**What did you already try?**
i *exactly* followed the instructions for installing octodash and the guide for the hyperpixel4 on a fresh install of octopi/octoprint. i also did `export DISPLAY=:0` and `ratpoison` where ratpoison gave a `Segmentation fault` as its output. 

**General Information:**
 - Raspberry Pi 4b with Hyperpixel4 rectangular screen.
 - octopi v1.0.0cam
 - OctoDash V2.3.1
 - OctoPrint V1.10.1 (new camera stack)

**Additional context**
Add any other context or screenshots about the feature request here.

ItsCharged, 2024-06-18:
> **What doesn't work?** the screen displays nothing when given power. **What did you already try?** i _exactly_ followed the instructions for installing octodash and the guide for the hyperpixel4 on [...]

So that means you also tried installing drivers?
Try to disable the v3d drivers in config.

wangyao127, 2024-06-20:
[image]

wangyao127, 2024-06-20:
Raspberry Pi 4b with Hyperpixel4 rectangular screen.
octopi v1.0.0cam
OctoDash V2.3.1
OctoPrint V1.10.1 (new camera stack)

wangyao127, 2024-06-20:
Raspberry Pi 4b with Kedei LCD 35 HDMI screen.
octopi v1.0.0cam
OctoDash V2.3.1
OctoPrint V1.10.2 (new camera stack)

patritha, 2024-06-21:
@ItsCharged how do i do that? sorry if thats a dumb question but im bot good with the pi

ItsCharged, 2024-06-21:
SSH into your pi. Navigate to the config by `cd ~/boot/` 
Or `cd ..` and again `cd ..` and last time `cd ..` to make sure youre in the right Directory. Then do `cd boot` and `sudo nano config.txt`. Now search for a line that says `dtoverlay=vc4-kms-v3d` and put a hashtag in Front of it like this `#dtoverlay=vc4-kms-v3d` Press Ctrl + x press y press enter and execute `sudo reboot -f`. Youre welcome. Tell me if it works or doesent.

patritha, 2024-06-21:
OMG it works thank you so much!!! im so happy now

closed 2024-06-21 by patritha