UnchartedBull/OctoDash #4612: OctoDash Segmentation Fault on Raspberry Pi 5 OctoPi 1.1.0

autronix, 2024-11-29:
**What doesn't work?**
When I run octodash, it throws a `Segmentation Fault`
[image]

**What did you already try?**
- I followed all the instructions on the troubleshooting guide
- installed rpi additional ui packages
- I edited the .xinit file to only run `ratpoison` - ratpoison starts, I can interact with keyboard and touch
[image]
- Tried starting Octodash from `rapoison` doing `^C !` then entering `octodash` - Segfaults
- Tried running Octodash from a `ratpoison` shell - Segfaults
- Tried disabling .xinitrc file - starts the window manager and tried running the Octodash command - SegFaults
- Tried to do a fresh install of OctoPi, and install OctoDash through the script - Same issue
- Checked and tried several other issue reports and forum posts associated with the screen, rpi and octodash - no solution

**General Information:**
 - Raspberry Pi 5 4Gb with Waveshare 5" DSI screen (connected to DSI 1)
 - OctoPi 1.1.0
 - OctoDash Version 2.3.1
 - OctoPrint Version 1.10.3

**Additional context**
Notes:
- OctoPi was installed from a nightly image since the Raspberry Pi imager does not support OctoPrint on RPi 5
- OctoPrint seems to be working fine, along with the webcam

I'm not sure what else I should be trying. I can't seem to find any logging information relating to Octodash that would provide me with more details about the error.

From what I can tell, xinit and startx are working, including the touch functionality; I'm able to start both `ratpoison` and the window managers. 

Any help appreciated.

waysert, 2024-11-29:
Hi, i am facing the same challenge. I already tried pretty the same possible solutions but nothing worked in my case. I really hope there is an solution out there for this issue. I really want to get this running, if anybody has an idea, i can test it, i run a second Installation where i can test new solutions without destroying my running OctoPrint installation ;-)

waysert, 2024-11-30:
I'm speechless, it works also for me. i used the original install script from OctoDash, made the changes you mentioned in the config.txt, and set manually the boot option B2 (Console with Autologin). Now it runs!! Thank you for this huge step forward!!

closed 2025-03-18 by queengooborg