UnchartedBull/OctoDash #4710: Rotate Screen produces weird graphical results

RickSeiden, 2025-02-10:
I rotated my screen using the instructions found [here](https://github.com/UnchartedBull/OctoDash/wiki/Rotate-screen).  The screen did rotate, but some things are weird.

This is what it normally looks like when I press the Files icon and then select the Local folder:

[image]

[image]

Note that the individual files have their thumbnails.

When the screen is rotated, I press the Files icon and then select the Local folder this is what I see:

[image]

[image]

The font is smaller, the thumbnails are missing, and there is a weird artifact at the top of the screen.

**What did you already try?**
I followed the instructions on the [Rotate Screen](https://github.com/UnchartedBull/OctoDash/wiki/Rotate-screen) instructions.

**General Information:**
 - Raspberry Pi 3B+
 - Waveshare 4.3 inch capacitive touchscreen 800x480
 - Linux octopi 5.15.30-v7+ #1535 SMP Wed Mar 23 16:38:23 GMT 2022 armv7l
 - OctoPi version: 0.17.0
 - OctoPrint version: 1.10.3
 - OctDash version: 2.3.1

**Additional context**
Everything functions, it's just not displaying correctly.  The font is smaller, that artifact is there when looking at folders and the thumbnails are missing, are the big three things.

Thank you!

RickSeiden, 2025-02-10:
I just noticed that in the "normal" mode picture above, it's my Local folder, but in the "rotated" mode picture it's my SD card.  That might explain the thumbnails, but not that artifact at the top of the screen or the smaller font.

queengooborg, 2025-03-18:
At this time, OctoDash requires landscape orientation and is not designed to be run on a portrait orientation screen.  The guide you reference is not particularly meant for setting the screen to landscape, but rather for fixing a screen in the wrong orientation.

I've opened #4801 to talk about using sizing techniques that will be more suitable for varying screen sizes.

RickSeiden, 2025-03-18:
Thanks for looking into it.  I would only recommend that you update the article to say it's not intended to put OctoDash into portrait mode. Other than that, knowing that it doesn't work because it wasn't designed to work that way is enough for me.

closed 2025-03-18 by RickSeiden

queengooborg, 2025-03-18:
I had opened #4802 to update the readme accordingly -- in the meantime, I'm also considering adding portrait screen support in v3.0.0 down the line!