UnchartedBull/OctoDash #4723: OctoDash is not starting automatically

Gargoyle1212, 2025-02-14:
I cannot get OctoDash to auto start to the display.

I have tried the manual install and the auto install.  I have also done the raspi-config tool to make sure it was set to autostart.  I have reboot the whole pi many times and it still only shows the command line for user name.  

**General Information:**
 Raspberry Pi 4 4GB with a 4in touch screen display
 I am running Octopi to have OctoPrint for my Ender 3 Pro
 - OctoDash v0.0.7 that is what it shows for OctoDash Companion
 - OctoPrint v1.10.3

Gargoyle1212, 2025-02-14:
I tried the below and still a no go.

pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ octodash
-bash: octodash: command not found
pi@octopi:~ $

jneilliii, 2025-02-14:
the companion plugin is separate from the OctoDash application. based on command not found, it sounds like it's not installing at all. Try to run this installer script and see if it's any better. 

```
bash <(wget -qO- https://github.com/jneilliii/OctoDash/raw/main/scripts/install.sh)
```

Gargoyle1212, 2025-02-14:
I used your link and it still didn't auto start.

Here is what I got on the install through putty

OctoPrint version : 1.10.3
OctoPi version    : 1.0.0

pi@octopi:~ $ bash <(wget -qO- https://github.com/jneilliii/OctoDash/raw/main/sc  ripts/install.sh)
Installing OctoDash v2.3.1, armhf
Installing Dependencies ...
[sudo] password for pi:
All packages are up to date.
bc is already the newest version (1.07.1-2+b1).
desktop-file-utils is already the newest version (0.26-1).
libappindicator3-1 is already the newest version (0.4.92-8).
libatspi2.0-0 is already the newest version (2.38.0-4+deb11u1).
libavahi-compat-libdnssd1 is already the newest version (0.8-5+deb11u3).
libnotify4 is already the newest version (0.7.9-3).
libnss3 is already the newest version (2:3.61-1+deb11u4).
libsecret-1-0 is already the newest version (0.20.4-2).
libuuid1 is already the newest version (2.36.1-8+deb11u2).
libx11-dev is already the newest version (2:1.7.2-1+deb11u2).
libxss1 is already the newest version (1:1.2.3-1).
libxtst6 is already the newest version (2:1.2.3-1).
ratpoison is already the newest version (1.4.9-1).
x11-xserver-utils is already the newest version (7.7+8).
xdg-utils is already the newest version (1.1.3-4.1).
xinit is already the newest version (1.4.0-1).
xserver-xorg is already the newest version (1:7.7+22).
libgtk-3-0 is already the newest version (3.24.24-4+rpt12+deb11u4).
libpam0g-dev is already the newest version (1.4.0-9+deb11u1+rpt2).
The following package was automatically installed and is no longer required:
  raspinfo
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
? Which plugins should I install (you can also install them via the Octoprint UI   later)? (Press <space> to select, <enter> to finalize)                        O  ctoDash Companion, Display Layer Progress, Filament Manager, Preheat Button, Pri  nt Time Genius
Installing Plugins...
Installing OctoDash v2.3.1, armhf ...
octodash.deb         100%[====================>]  59.89M  3.97MB/s    in 15s
Selecting previously unselected package octodash.
(Reading database ... 62024 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (2.3.1) ...
Setting up octodash (2.3.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for mailcap (3.69) ...
? Should I setup OctoDash to automatically start on boot? yes
yes
Setting up Autostart ...
Setting up Console Autologin ...
Setting Permissions ...
OctoDash will start automatically on next reboot. Please ensure that auto-login is enabled!
? Should I setup the update script? This will allow installing '~/tmp/octodash.deb' without sudo or root access. For more info visit the Update section of the wiki.  (Use arrow keys)                                                               yes
Setting up update script ...
? Shall I reboot your Pi now? yes
OctoDash has been successfully installed! :)

Gargoyle1212, 2025-02-14:
After install with you link I tried to run octodash through putty and got the below.  The screen still shows the login prompt.

pi@octopi:~ $ octodash
selected language: en
Trace/breakpoint trap
pi@octopi:~ $

Gargoyle1212, 2025-02-14:
I ran the raspi-config tool and selected to autologin and it rebooted and still did not autostart

Gargoyle1212, 2025-02-15:
Got it auto start but now I have a new problem and if I can't figure it out will start a new issue.

Fixed by doing a whole new install and did no plugins or anything until I had octodash installed.

closed 2025-02-15 by Gargoyle1212