UnchartedBull/OctoDash #4726: OctoDash Can't connect to Octoprint

Gargoyle1212, 2025-02-15:
I am setting up OctoDash on my Raspberry Pi running Octoprint and can't get past the setup screens on the OctoDash for the API portion.

**What did you already try?**
I have gotten it to start screens, and it is trying to authenticate it for the API key.  It keeps saying iot can't connect even after multiple restarts.  I found where i have to now manually add it in Octoprint and named it OctoDash and it generated a key.  I typed that key in manually to cintune and when it got to the end it still couldn't connect and gave me the unknown 0 error.  I have tried everything I can find on the web but it seems to be all outdated.

**General Information:**
 - Raspberry Pi 4 Model B 4gb with 4 in display touch screen.
 - OctoPi Build 2024.11.05.092907 with "webcamd"
 - OctoDash v0.0.7 (don't know if this is correct)
 - OctoPrint v1.10.3

[image]

jneilliii, 2025-02-15:
Did you try with the server address http://localhost:5000/?

Gargoyle1212, 2025-02-16:
Manual setting in the web browesr portion on the Pi?  In the Pi I only have the options for send request or to type in an API key.

jneilliii, 2025-02-16:
No, in the OctoDash settings/setup screen.

Gargoyle1212, 2025-02-18:
ok I can't find where that is.  I manually entered my API key and can't get past the setup screens.  It stops at the final screen with a 0 unknown error.

[image]

Gargoyle1212, 2025-02-19:
I got it working.

I removed OctoDash Companion plugin
Then ran the below commands through SSH
          rm ~/.config/octodash/config.json
          sudo service getty@tty1 restart

then go through the OctoDash setup again, when you get to the option to select the OctoPrint url choose to use manual method and enter http://localhost:5000/

closed 2025-03-18 by queengooborg