UnchartedBull/OctoDash #4849: /plugins must have required property 'companion'

patritha, 2025-03-20:
**Describe the bug**
after updating it wanted me to update the config but when getting to the plugins screen i cant select anything and when clicking next it has the error in the title along with stuff about tuya and wemo, wich are plugins i didnt even select

**To Reproduce**
Steps to reproduce the behavior:

1. Go to update
2. get to update config menu
3. press next after being unable to select plugins
4. See error

**Expected behavior**
its supposed to allow me to select plugins and exit the configuration

**Screenshots**
[image]

**General Information:**

- Hardware raspberry pi 4b + hyperpixel 4
- OS Info octopi v1 new camera stack
- OctoDash Version 2.4.0
- OctoPrint Version 1.10.3
- python 3.9.2

**Additional context**
using nox theme

queengooborg, 2025-03-21:
This should be fixed in #4848!

closed 2025-03-21 by queengooborg

patritha, 2025-03-21:
awesome sauce