UnchartedBull/OctoDash #4873: Distribute as OctoPrint plugin instead of electron bundle

Hillshum, 2025-03-22:
I propose chaning the way this project is distributed. Instead of packaging as an Electron application bundled into a `.deb` package, this should be distributed as a plugin for OctoPrint, then run in a simple kiosk browser

This has the following advantages:
* Streamlined path for distributing updates
* Streamlined path for instalation
* Support for other device types (phones/tablets, other SBCs, things not Debian-based...)

I can't think of any long-term disadvantages, just some development work required:
* Migrating config to be handled on the OctoPrint side (this has the side-effect of capturing that config in native backups
* Handling API keys (probably storing it in browser localstorage?)
* Bundling/plugin stuff
* Setting up the local browser

Am I missing anything, especially on the side of dev work and any other downsides?

Hillshum, 2025-03-22:
One downside is running commands (like reboot) on the OctoDash host would be lost if running on a standard browser. That may not be such a concern, though

jneilliii, 2025-03-22:
I'd be curious to how many people actually run OctoDash separately/remote from the same host. I assume it's rare, and in the case of using a tablet/phone, etc. if that device needs to be rebooted easy enough to do manually I would think.