UnchartedBull/OctoDash #4875: Update to 2.4.1 and updated Nox custom-styles.css fills drive

morphias2004, 2025-03-22:
**Describe the bug**
Updated to v2.4.1 and updated the custom-styles.css for the Nox these to the new version and rebooted.

Within 10mins, the hard drive had been completely filled till 0% was remaining.

Found that /var/log/syslog and kern.log had grown to over 12G.

[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Updated to 2.4.1
2. Import Nox custom-styles.css
3. Reboot

Happened on two completely separate installations.

**Expected behavior**
For it not to fill the log files in minutes.

**Screenshots**

[image]

**General Information:**

RPi4

- Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
- OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
- OctoDash Version [e.g. v1.0.0]
- OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context about the problem here.

morphias2004, 2025-03-22:
With the assistance of @jneilliii I have downgraded to 2.3.1, which has resolved the issue.

Hillshum, 2025-03-22:
Did you apply the Nox styles the second time around?

morphias2004, 2025-03-22:
I think the Nox update was a red herring. It just happened to be updated around the time the logs had filled the disk.

Subsequent reinstalls of 2.4.1 had the issue without the Nox theme being installed.

I haven't applied the Nox theme since the roll back.

Hillshum, 2025-03-22:
Yeah, sounds like a red herring

Hillshum, 2025-03-26:
So we've updated Electron in later builds. Can you try running one of these? https://github.com/UnchartedBull/OctoDash/actions/runs/14010107574/artifacts/2801222591 (you'll have to be logged-in to Github, so you may need to download, then use `scp` or some such to copy to your Pi)

Also, you may be able to get more useful logs by adding `2>&1 > $HOME/octodash.log` to end of the `octodash` command in your `.xinitrc` file

Hillshum, 2025-04-02:
I did encounter this the other day, I think with 2.4.0. Not sure what kicked it off.

morphias2004, 2025-04-12:
Apologies for the delay in coming back to you on this. I just updated to 2.4.2 today and the issue has not reoccurred.

I think this can be closed now.

pooley182, 2025-04-22:
I just did an install of OctoDash 2.4.2 on top of OctoPi (Installed from Raspberry Pi Imager) and am seeing this same issue with the drive being filled after OctoDash starts with the same `Unhandled prefetch abort: breakpoint debug exception (0x222) at 0x00000000` log line filling `kern.log` and `syslog`

- Raspberry Pi 4 Model B Rev 1.4
- Raspbian 11 (bullseye) - OctoPi
- OctoPrint v1.10.3