UnchartedBull/OctoDash #4931: NOX theme is messed up while printing

patritha, 2025-04-07:
**Describe the bug**
during a print, the interface looks all jumbled up with the nox theme

**To Reproduce**
Steps to reproduce the behavior:

1. install nox theme
2. start print
3. see issue

**Expected behavior**
it to look like it used to

**Screenshots**

[image]

**General Information:**

- Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
- OS Info Octopi 1.0.0cam
- OctoDash Version 2.4.2
- OctoPrint Version 1.10.3
- Screen: 800x480 (Hyperpixel 4)

**Additional context**
Add any other context about the problem here.

jneilliii, 2025-04-08:
Did you re-upload the css file after installing the updated version or is this a fresh install?

patritha, 2025-04-08:
reuploaded the css

jneilliii, 2025-04-08:
it may have been updated since the 2.4.2 release, can you try the one from that tag release? 

https://github.com/UnchartedBull/OctoDash/blob/v2.4.2/themes/NOX/custom-styles.css

jneilliii, 2025-04-08:
Just tried that version and it doesn't appear to make a difference.

jneilliii, 2025-04-08:
wonder if the old 2.3.1 version still works. 

https://github.com/UnchartedBull/OctoDash/blob/v2.3.1/themes/NOX/custom-styles.css

patritha, 2025-04-09:
just uploaded and tried it and it didnt change anything unfortunitely

Mackalope, 2025-06-09:
Any solution paths?

jneilliii, 2025-06-09:
on 640x480 resolution, this doesn't look so bad, but not perfect, remove the txt extension.

[custom-styles.css.txt](https://github.com/user-attachments/files/20647609/custom-styles.css.txt)

Basically it has this in it.

```
.printer-status {
    position: absolute;
    top: 27vh !important;
    left: 35vw !important;
    width: fit-content !important;
}
```

closed 2025-07-08 by queengooborg