UnchartedBull/OctoDash #4946: Unable to Display OctoDash

pigaro, 2025-04-28:
I am attempting to update my Raspberry Pi 4B boot SD from Debian Buster with Python 3.7 to Debian Bullseye with Python 3.9 using the OctoPi image with the Raspberry Pi Imager. Writing the SD completed without issue.

I updated Bullseye to the current version (apt update, apt upgrade, apt full-upgrade). Installed Samba. Rebooted. Installed Klipper manually and installed KIAUH. Copied the Klipper printer.cfg file to the pi home directory. Then I opened OctoPrint and restored the configuration from a backup. All completed without issues.

I used the automatic script to install OctoDash by copying from the web site and pasting into the pi ssh session. The install completed with no warnings or errors. I selected to not install any plugins, to automatically start on boot, and to allow the update scripts to operate without special permissions, and then rebooted the Pi.

On rebooting the display showed the usual console messages, then went blank, and then showed a flashing box indicating No Signal.

I took the following troubleshooting steps:

Double checked B2 Console Autologin Text console, automatically logged in as ‘pi’ user. Rebooted.
     No change
Display in use is HDMI
No special settings or driver installed
Reconfigured X11 server using sudo dpkg-reconfigure x11-common. Rebooted.
     No change
Installed additional packages using sudo apt install libpam0g-dev libx11-dev
     Nothing upgraded, installed, or removed.
Installed more packages using sudo apt install raspberrypi-ui-mods
     One warning: (zenity:9490): Gtk-WARNING **: 23:37:51.087: cannot open display: 
     Rebooted. No change.
Allowed any user to use init
     No change
Environment variable DISPLAY is not defined, added export DISPLAY=:0 to .bash_profile
     Rebooted. Verified DISPLAY was set. No change.
Installed desktop environment using sudo /home/pi/scripts/install-desktop
      No change
Attempted to run ratpoison
     Segmentation fault
Attempted to run octodash
     Result: selected language: en
     No change on display

In a previous attempt, before defining the DISPLAY environment variable, the following was tried:
pi@octopi:~ $ octodash
selected language: en
[1323:0427/185225.544430:ERROR:ozone_platform_x11.cc(246)] Missing X server or $DISPLAY
[1323:0427/185225.544547:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
Segmentation fault

pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ ratpoison
Segmentation fault

I am not sure where to look for logs.

System information:
Raspberry Pi 4B with 4 GB memory
Image created from Octopi 1.0.0 with OctoPrint 1.11.0 with new camera stack
Klipper v0.13.0-51
Waveshare 5 inch HDMI touchscreen display, 800 x 480

pigaro, 2025-05-10:
I built a new image using Octopi 1.1.0 with OctoPrint 1.11.0 and the classic camera stack. This version has Raspbian Bookworm with Python 3.11.2. I have installed Klipper and Samba. I was able to restore my OctoPrint backup and verify that I can control the printer movement and heaters (meaning Klipper is working).

After completing the install and allowing the OctoDash script to reboot the Raspberry Pi, the display remained blank with the periodic "No Signal" message.

I tried the following commands:
pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ ratpoison
Can't open display

Attached is a file with a copy of the messages during the install. I saw one warning and no errors.

[OctoDash install messages.txt](https://github.com/user-attachments/files/20132841/OctoDash.install.messages.txt)

pigaro, 2025-05-22:
I was able to get OctoDash to work on Raspbian Bullseye. I started with a lite Bullseye image, manually installed OctoPrint, installed Klipper using KIAUH, and then installed OctoDash using the automated script. OctoDash worked with Bullseye, but not Bookworm.

queengooborg, 2025-07-07:
This is sounding like the same issue that was reported in #5065, which I have just fixed.  The install script _should_ now work on Bookworm (OctoPi or otherwise), if you're down to give it a whirl!

pigaro, 2025-07-07:
Queengooborg,

You wrote:
> This is sounding like the same issue that was reported in #5065 <https://github.com/UnchartedBull/OctoDash/issues/5065>, which I have just fixed. The install script should now work on Bookworm (OctoPi or otherwise), if you're down to give it a whirl!
> 
OK, thanks for letting me know. I will give it a try and let you know.

Lee Mendoza
***@***.***

pigaro, 2025-07-07:
Queengooborg,

The fix to the install script worked like a charm. Thanks for the fix and thanks for letting me know. 

Lee Mendoza
***@***.***

closed 2025-07-07 by pigaro