UnchartedBull/OctoDash #5065: Installation doesn't create the config.json file

lluisd, 2025-06-23:
**Describe the bug**
After running the installaction script the .config/octodash folder is empty
[image]

**To Reproduce**
Install last version using the script provided by octodash

**General Information:**

- Hardware: raspberry pi 3b+
- OctoPrint 1.11.2 Python 3.11.2 OctoPi* 1.1.0

jneilliii, 2025-06-23:
I'm pretty sure that's normal. The file gets created during the initial setup in the app once it's ran.

lluisd, 2025-06-24:
> I'm pretty sure that's normal. The file gets created during the initial setup in the app once it's ran.

how can i run the app? When i run octodash from ssh it says:

```
lluis@raspberrypi:~ $ octodash
selected language: en
[1669:0624/161703.387226:ERROR:ozone_platform_x11.cc(246)] Missing X server or $DISPLAY
[1669:0624/161703.387515:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
Segmentation fault
```

It was working fine in the past with the v2.3.1 and using octopi 1.0.0 but I upgraded to octopi 1.1.0 and now I cannot make it work. with the octopi 1.0.0 I couldn't upgrade from the screen so I finally reflashed my sd card.

queengooborg, 2025-07-07:
Well, OctoDash _should_ be launching on your Raspberry Pi on bootup.  If you need to launch it from SSH for troubleshooting, you can run `export DISPLAY=:0` before running `octodash`!

PR #5087 (2025-07-07): Fix autoboot setup process on Debian 12 / OctoPi 1.1.0
This fixes #5065.
closed 2025-07-07 by queengooborg