UnchartedBull/OctoDash #5108: Ocotdash not working anymore with FYSETC-CTP40 with: no screens found(EE)

lluisd, 2025-07-09:
**Describe the bug**
 no screens found(EE)

To Reproduce
Install last version using the script provided by octodash on a fystec-ctp40

General Information:

Hardware: raspberry pi 3b+
OctoPrint 1.11.2 Python 3.11.2 OctoPi* 1.1.0

it was working fine in the past with the v2.3.1 and using octopi 1.0.0

some logs:
```
[   125.211]
X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
[   125.222] Current Operating System: Linux raspberrypi 6.12.25+rpt-rpi-v7 #1 SMP Raspbian 1:6.12.25-1+rpt1 (2025-04-30) armv7l
[   125.222] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=2e541531-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=ES
[   125.238] xorg-server 2:21.1.7-3+rpt3+deb12u9 (https://www.debian.org/support)
[   125.240] Current version of pixman: 0.44.0
[   125.243]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   125.243] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   125.250] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Jul  9 16:28:48 2025
[   125.252] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   125.254] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   125.255] (==) No Layout section.  Using the first Screen section.
[   125.255] (==) No screen section available. Using defaults.
[   125.255] (**) |-->Screen "Default Screen Section" (0)
[   125.255] (**) |   |-->Monitor "<default monitor>"
[   125.256] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[   125.256] (**) |   |-->Device "kms"
[   125.256] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   125.256] (**) Option "Debug" "dmabuf_capable"
[   125.256] (==) Automatically adding devices
[   125.256] (==) Automatically enabling devices
[   125.256] (==) Automatically adding GPU devices
[   125.256] (==) Automatically binding GPU devices
[   125.256] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   125.256] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   125.256]    Entry deleted from font path.
[   125.256] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[   125.257] (==) ModulePath set to "/usr/lib/xorg/modules"
[   125.257] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[   125.257] (II) Loader magic: 0x215080
[   125.257] (II) Module ABI versions:
[   125.257]    X.Org ANSI C Emulation: 0.4
[   125.257]    X.Org Video Driver: 25.2
[   125.257]    X.Org XInput driver : 24.4
[   125.257]    X.Org Server Extension : 10.0
[   125.260] (--) using VT number 7

[   125.261] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   125.261] (II) no primary bus or device found
[   125.261] (II) LoadModule: "glx"
[   125.262] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   125.265] (II) Module glx: vendor="X.Org Foundation"
[   125.265]    compiled for 1.21.1.7, module version = 1.0.0
[   125.265]    ABI class: X.Org Server Extension, version 10.0
[   125.265] (II) LoadModule: "modesetting"
[   125.266] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[   125.399] (II) Module modesetting: vendor="X.Org Foundation"
[   125.399]    compiled for 1.21.1.7, module version = 1.21.1
[   125.399]    Module class: X.Org Video Driver
[   125.399]    ABI class: X.Org Video Driver, version 25.2
[   125.399] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   125.406] (WW) Falling back to old probe method for modesetting
[   125.406] (EE) open /dev/dri/card0: No such file or directory
[   125.406] (WW) Falling back to old probe method for modesetting
[   125.406] (EE) open /dev/dri/card0: No such file or directory
[   125.406] (EE) No devices detected.
[   125.406] (EE)
Fatal server error:
[   125.406] (EE) no screens found(EE)
[   125.406] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[   125.406] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   125.406] (EE)
[   125.425] (EE) Server terminated with error (1). Closing log file.
```

queengooborg, 2025-07-09:
Do you happen to know if it was working on OctoPi 1.0.0 with a newer version of OctoDash?

stoff88, 2025-08-01:
I have the exact same problem on the latest setup.
I use 

- Latest OctoPie 1.1.0 with OctoPrint 1.11.2.
- Adafruit PiTFT Plus 480x320 3.5" Resistive Touchscreen for Raspberry Pi, installed with latest driver according to Adafruit.
- Installed OctoDash 2.5.3.

I have tried to just write the image and proceeded without apt upgrade, but i have also tried to do apt upgrade first after OctoPie image are writed, then proceeded with screen driver and OctoDash.

lluisd, 2025-09-05:
still an issue for 2.6.1