UnchartedBull/OctoDash #5109: The setup wizard on the latest version fails on Validating Config after update

patritha, 2025-07-10:
**What doesn't work?**
The setup wizard fails at the end on "Validating Config". the errors are `/plugins/<plugin name> must NOT have additional properties`. 

**What did you already try?**
idk i rebooted many times and updated again after version 2.5.2 came out over the console.

**General Information:**

- Hardware: Raspberry Pi 4b, Hyperpixel 4
- OS: Info Octopi 1.0.0cam
- OctoDash Version: v2.5.2
- OctoPrint Version: v1.11.2

**Additional context**
[image]

queengooborg, 2025-07-10:
What version of OctoDash were you running before?

patritha, 2025-07-10:
it was working fine on 2.4.2, i updated to 2.5.1 where it started doing this, then i updated to 2.5.2 and nothing changed

PR #5110 (2025-07-10): Fix configuration migration for plugins
Follow-up to #5095 which didn't actually fix the migration.  Fixes #5109.