UnchartedBull/OctoDash #5163: OctoDash won't start after update, logo shown only

C0untZero, 2025-07-19:
**Describe the bug**
I've upgraded from 2.4.2 to 2.5.0 using the menu controls, and then to 2.5.3 using the upgrade bash script.
Screen now shows only OctoDash logo and goes no further. 
I've had the config migration error when I upgraded to 2.5.0.

**Expected behavior**
OctoDash starts normally after update.

**General Information:**
- RPi 3B v1.3
- OctoPi/Raspbian buster 5.10.103-v7+
- OctoDash 2.5.3
- OctoPrint 1.11.1

Where can I find some additional logs to provide?

jovanzlatanovic87, 2025-07-22:
Just a line saying: "selected language: en" and then no end to the command

C0untZero, 2025-07-23:
Yup, that did it. I had to do the initial configuration again, but after that when I diff the two config files I don't see a significant difference between them apart from some element ordering. Could that be the cause? You guys want me to post the config files?

queengooborg, 2025-07-23:
Ordering shouldn't cause any issues, but if you're down to upload the config file, that would be helpful for debugging!